gohai/processing-glvideo #2: Video offset from sketch window and forced to "width" and "height"

SciFun, 2016-05-13:
I  was able to run this on a Pi-3, but the video is offset from the sketch window and has a size of "width" and "height" that I could not change. This is a great first step! Finally able to use processing on the Pi with video. Can any other size video other the the sketch size() be used? how?

gohai, 2016-05-24:
@SciFun Could you elaborate? Which example were you trying?

SciFun, 2016-05-27:
My apologies. It is the simple capture example.
used in this manner, example code only slightly modified:

import gohai.glvideo.GLVideo;
GLVideo video;

void setup() {
  size(640, 480, P2D);
  // this will use the first recognized camera
  video = new GLVideo(this, "v4l2:///dev/video0");
  video.play();
}

void draw() {
  if (video.available()) {
    video.read();
  }
  image(video, 40, 40, 320, 240); // try to get the video back inside the sketch window
}

// a little bit off of the sketch window, but it works. The image() offset was ignored

SciFun, 2016-05-27:
I noticed that it is really the video background that is misplaced and also a second cursor appears. The image() function is working fine but somehow the rendering of the sketch window is offset from the frame. Maybe a P2D issue?

SciFun, 2016-05-27:
So I ran several P2D and P3D examples (Planets, yellowtail, etc) and anything that uses P2D or P3D is offset up and to the left by ~50 pixels in x and y. So the SimpleCapture example is not the issue. For some reason the rendering is offset ini P3D and P2D (at least on my Pi-3 using HDMI). Sketches that do not use P3D or P2D are fine but GLRenderer requires them so the SimpleCapture example requires that. Sorry for the slow and poor communication on this. Thanks for getting processing running on the Pi..

SciFun, 2016-05-27:
So I found that I could change the P2D and P3D offset so that the display was back inside the sketch window by setting the overscan_left=0 and overscan_top=0 variables in the config.txt (to zero) but of course that messed the desktop up. That indicates to me that P3D and P2D do not compensate for the overscan values. Also noticed that displayHeight indicates 1920 (displayWidth) when it should be 1080 for the mode I'm using (16).

gohai, 2016-05-28:
@SciFun What happens if you use `disable_overscan=1` instead?

But in any way, not a GL Video issue, and also likely nothing we can do anything about in Processing or even JOGL (the Java GL library), but instead an issue in the (closed source) GLES2 driver. If you want me to follow up with this, please file a bug for the main Processing project, that includes a photo of what you see when you run one of the P3D example sketches. Also include your `config.txt` and information about your display.

closed 2016-05-28 by gohai