gohai/processing-glvideo #4: Loops and Exectutables

cosmiccosmo, 2016-05-25:
HI,
first of all thank you for great work for the community!! I just installed your new Rasbian image with processing and tested your glvideo library on Pi02 and Pi03 and discovered following problems:

1. I used very short mp4 sequences and discovered that they stop approximatly a second before they really end. If they are looped the sequence is paused for this time until it starts in the loop again. Unfortunately the sequence is never played in full length.

2. If I try to create a executable, processing informs me that it failed to create a linux 32bit and 64bit version. That is nothing to wonder about, but it also fails to create the arm version!?

gohai, 2016-05-25:
@cosmiccosmo Thanks for your report.

I haven't noticed this behavior, but I'll have a look. The popups are expected, since the library currently only exists for Linux on ARM. You should be able to create an executable if you export it from a Windows or Mac machine. (Make sure you use the latest version from the Contribution Manager though.) I'll look into why it doesn't work on ARM itself. Haven't done that myself.

gohai, 2016-05-27:
@cosmiccosmo I tried reproducing your issue with regards to export but things work fine here..

My steps:
Download Processing 3.1.1
Install GL Video library
Open up `SingleVideo` example
`File` > `Export Application...`
Checked the `Linux` checkbox
Clicked through two popups telling me about no native library being available for (Intel) 32 and 64
Waited a few seconds
Directory appears

Can you help me here?

cosmiccosmo, 2016-05-27:
Thank you, Gohai! Sunday evening I will have time to check, if it might be a problem of the raspian image or if I just could not wait long enough! In my eager state of mind that evening last thing might be possible as well! Were you able to look into the other problem as well?

gohai, 2016-05-27:
@cosmiccosmo I have some idea why the second issue might be happening. Will look into it and do something about it, but it won't be soon. If you need looping videos earlier, perhaps you could just create a longer video file with a couple of iterations chained together in the mean time?

cosmiccosmo, 2016-05-29:
@gohai 
my export problem was due to free sdcard space. i am sorry about that one!

gohai, 2016-05-29:
Thanks for checking @cosmiccosmo!