gohai/processing-glvideo #7: Windows support

gohai, 2016-08-10:


codeanticode, 2016-10-08:
Chatting with Matt about this, he says that we need to do the following operations:

* grab the opengl context from jogl as it is currently done (https://github.com/gohai/processing-glvideo/blob/master/src/native/impl.c#L338)
* wrap the context  right before this line:
https://github.com/gohai/processing-glvideo/blob/master/src/native/impl.c#L162
not where it is done for other platforms (https://github.com/gohai/processing-glvideo/blob/master/src/native/impl.c#L384) bc, on Windows, the wrapping has to happen when the OpenGL is NOT current.

The [QT plugin](https://cgit.freedesktop.org/gstreamer/gst-plugins-bad/tree/ext/qt/gstqtglutility.cc) can be useful as reference, for example [this comment](https://cgit.freedesktop.org/gstreamer/gst-plugins-bad/tree/ext/qt/gstqtglutility.cc#n181) explains the reason why doing the context wrapping when it is not current.

Mostly notes to self :-)

Andres

gohai, 2018-04-16:
I have decided to make the library Raspberry-Pi only, because I don't have enough time to deal with other platforms. Closing this bug for this reason.

closed 2018-04-16 by gohai