gohai/processing-glvideo #8: Capture enumeration

gohai, 2016-08-10:


codeanticode, 2016-10-11:
[GstDeviceProvider](https://gstreamer.freedesktop.org/data/doc/gstreamer/head/gstreamer/html/gstreamer-GstDeviceProvider.html) is the relevant API to programatically list connected devices. Implementations are being created for different platform, see for example [GstDeviceProvider implementations for Windows](https://bugzilla.gnome.org/show_bug.cgi?id=747757). It was mentioned during the conference that patches are waiting to be merged with equivalent implementations for OSX.

gohai, 2016-10-12:
Just implemented: https://github.com/gohai/processing-glvideo/tree/wip-list-devices

gohai, 2016-10-12:
This is the information that `wip-list-devices` is currently spitting out on Linux:

    FaceTime HD Camera (Built-in)
	class: Video/Source
	caps: video/x-raw, format=(string)YUY2, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:7:1, framerate=(fraction)10/1; video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:7:1, framerate=(fraction){ 2997/100, 25/1, 24/1, 15/1 }; video/x-raw, format=(string)YUY2, width=(int)480, height=(int)360, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:7:1, framerate=(fraction){ 2997/100, 25/1, 24/1, 15/1 }; video/x-raw, format=(string)YUY2, width=(int)352, height=(int)288, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:7:1, framerate=(fraction){ 2997/100, 25/1, 24/1, 15/1 }; video/x-raw, format=(string)YUY2, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:7:1, framerate=(fraction){ 2997/100, 25/1, 24/1, 15/1 }; video/x-raw, format=(string)YUY2, width=(int)176, height=(int)144, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:7:1, framerate=(fraction){ 2997/100, 25/1, 24/1, 15/1 }; video/x-raw, format=(string)YUY2, width=(int)160, height=(int)120, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:7:1, framerate=(fraction){ 2997/100, 25/1, 24/1, 15/1 }; image/jpeg, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)2:4:7:1, framerate=(fraction){ 2997/100, 25/1, 24/1, 15/1 }; image/jpeg, width=(int)1024, height=(int)576, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)2:4:7:1, framerate=(fraction){ 2997/100, 25/1, 24/1, 15/1 }; image/jpeg, width=(int)960, height=(int)544, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)2:4:7:1, framerate=(fraction){ 2997/100, 25/1, 24/1, 15/1 }; video/x-raw, format=(string)I420, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, framerate=(fraction){ 2997/100, 25/1, 24/1, 15/1 }; video/x-raw, format=(string)I420, width=(int)1024, height=(int)576, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, framerate=(fraction){ 2997/100, 25/1, 24/1, 15/1 }; video/x-raw, format=(string)I420, width=(int)960, height=(int)544, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, framerate=(fraction){ 2997/100, 25/1, 24/1, 15/1 }; video/x-raw, format=(string)I420, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, framerate=(fraction){ 2997/100, 25/1, 24/1, 15/1 }; video/x-raw, format=(string)I420, width=(int)480, height=(int)360, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, framerate=(fraction){ 2997/100, 25/1, 24/1, 15/1 }; video/x-raw, format=(string)I420, width=(int)352, height=(int)288, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, framerate=(fraction){ 2997/100, 25/1, 24/1, 15/1 }; video/x-raw, format=(string)I420, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, framerate=(fraction){ 2997/100, 25/1, 24/1, 15/1 }; video/x-raw, format=(string)I420, width=(int)176, height=(int)144, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, framerate=(fraction){ 2997/100, 25/1, 24/1, 15/1 }; video/x-raw, format=(string)I420, width=(int)160, height=(int)120, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, framerate=(fraction){ 2997/100, 25/1, 24/1, 15/1 }; video/x-raw, format=(string)YV12, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, framerate=(fraction){ 2997/100, 25/1, 24/1, 15/1 }; video/x-raw, format=(string)YV12, width=(int)1024, height=(int)576, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, framerate=(fraction){ 2997/100, 25/1, 24/1, 15/1 }; video/x-raw, format=(string)YV12, width=(int)960, height=(int)544, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, framerate=(fraction){ 2997/100, 25/1, 24/1, 15/1 }; video/x-raw, format=(string)YV12, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, framerate=(fraction){ 2997/100, 25/1, 24/1, 15/1 }; video/x-raw, format=(string)YV12, width=(int)480, height=(int)360, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, framerate=(fraction){ 2997/100, 25/1, 24/1, 15/1 }; video/x-raw, format=(string)YV12, width=(int)352, height=(int)288, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, framerate=(fraction){ 2997/100, 25/1, 24/1, 15/1 }; video/x-raw, format=(string)YV12, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, framerate=(fraction){ 2997/100, 25/1, 24/1, 15/1 }; video/x-raw, format=(string)YV12, width=(int)176, height=(int)144, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, framerate=(fraction){ 2997/100, 25/1, 24/1, 15/1 }; video/x-raw, format=(string)YV12, width=(int)160, height=(int)120, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, framerate=(fract [... cut ...]
* constructors like the existing, where you can specify width, height & framerate if wanted - as well as one where you just pass a the device name and GStreamer caps

One could perhaps do some nice heuristic where - by default the code would select the highest resolution that is at least 30 fps, or something like this. What does the Video library do here?

In any way, lemme know - I shall hack this together quickly, at least for Linux. (On the Pi w/ limited power it'd be nice to be able to specify a custom resolution and fps.)

codeanticode, 2016-10-14:
Currently, the Video.list() function returns a list of strings where each item corresponds to a unique combination of device name, resolution, and fps, something like:

```
[0] "name=FaceTime HD Camera,size=1280x720,fps=30"
[1] "name=FaceTime HD Camera,size=1280x720,fps=15"
[2] "name=FaceTime HD Camera,size=1280x720,fps=1"
[3] "name=FaceTime HD Camera,size=640x360,fps=30"
[4] "name=FaceTime HD Camera,size=640x360,fps=15"
[5] "name=FaceTime HD Camera,size=640x360,fps=1"
[6] "name=FaceTime HD Camera,size=320x180,fps=30"
[7] "name=FaceTime HD Camera,size=320x180,fps=15"
[8] "name=FaceTime HD Camera,size=320x180,fps=1"
[9] "name=FaceTime HD Camera,size=160x90,fps=30"
[10] "name=FaceTime HD Camera,size=160x90,fps=15"
[11] "name=FaceTime HD Camera,size=160x90,fps=1"
[12] "name=FaceTime HD Camera,size=80x45,fps=30"
[13] "name=FaceTime HD Camera,size=80x45,fps=15"
[14] "name=FaceTime HD Camera,size=80x45,fps=1"
```

You can use that string to initialize the Capture object:

```
String[] cameras = Capture.list();
cam = new Capture(this, cameras[0]);
```

I think that having list() returning only the list of camera names, and then configs(cameraName) the list of resolutions and fps for the given camera is more clear and easier to use.

gohai, 2016-10-15:
The `wip-list-devices` branch now has a `list()`, a `configs(deviceName)` as well as a constructor that takes a `deviceName`.

What's left: hooking the newly created `GstElement` up with the pipeline, being able to pass caps (or `width`/`height`/`fps`) to it, and any caps filtering/massaging that should be done.

gohai, 2016-10-16:
Pretty much done now in `wip-list-devices`.

gohai, 2016-11-12:
Support for this just got merged into master. Still nothing for macOS and Windows as far as I can tell.

gohai, 2018-04-16:
I have decided to make the library Raspberry-Pi only, because I don't have enough time to deal with other platforms. Closing this bug for this reason.

closed 2018-04-16 by gohai