gohai/processing-glvideo #9: Hang on Mac

gohai, 2016-08-10:


codeanticode, 2016-10-08:
Hey @gohai it turns out if you call jump() only after the video becomes available, then there is no hang:

```
import gohai.glvideo.*;
GLMovie video1;
GLMovie video2;

boolean jumped = false;

void setup() {
  size(560, 203, P2D);
  video1 = new GLMovie(this, "launch2.mp4", GLVideo.MUTE);
  video2 = new GLMovie(this, "launch2.mp4", GLVideo.MUTE);  
  video1.loop();
  video2.loop();
}

void draw() {
  background(0);
    
  if (video1.available()) {    
    video1.read();
  }
  if (video2.available()) {
    if (!jumped) {
      video2.jump(2.0);
      jumped = true;
    }     
    video2.read();
  }
  image(video1, 0, 0, width/2, height);
  image(video2, width/2, 0, width/2, height);
}
```

Matt is looking into it. If he cannot find a solution, a cheap workaround could be to store in a stack all the calls requiring a wait_for_state_change when they are done before the stream is available, and actually issuing them as soon as the stream is ready. What do you think?

gohai, 2016-10-08:
@codeanticode we were aware of this earlier, but the question remains why wait_for_state_change doesn't work on Mac, and Mac only ... I think it's better if people could just call `jump` at any time - cause they will try to do so in any way ;)

gohai, 2016-10-08:
`wait_for_state_change` is also used in other parts, e.g. when the user is querying the length of the video file. We should be able to block until this information is available.

codeanticode, 2016-10-08:
Actually, we just realized that glvideo is creating a [new wrapped display](https://github.com/gohai/processing-glvideo/blob/master/src/native/impl.c#L386) for each movie, which is problematic because the cocoa display does funny things with CFRunLoop. Even though multiple videos appear to work in most situations (RPi, Linux, Mac w/out seek), creating multiple display is definitely not correct.

gohai, 2016-10-08:
Cool. Will try to implement real quick.

gohai, 2016-10-08:
Implemented in branch wip-global-display. Hang is still there in TwoVideos, though. Does this look right to you guys? Can I just re-used the first display as-is, or would I need to do something else (taking a reference, or similar)?

gohai, 2016-10-08:
https://github.com/gohai/processing-glvideo/commit/845df79b57c37705049c2f29c6621fabebae830d

codeanticode, 2016-10-08:
The wrapped context should not be reused, even though in practice it does not matter since each wrapped context wraps the same context, it is helpful for debugging to be able to differentiate between them.

gohai, 2016-10-08:
Ah, so global display, multiple contexts? Sec...

codeanticode, 2016-10-08:
yes

gohai, 2016-10-08:
Pushed as https://github.com/gohai/processing-glvideo/commit/2ad5b1fe2c7aba112e254e6f343c8a983038c3ce. (Issue remains.)

codeanticode, 2016-10-08:
not surprised, but it is still correct. Trying other things on CFRunLoop

gohai, 2016-10-08:
No - good catch! All for correctness and making sure what we're doing matches what the library expects us to do.

cansik, 2017-08-02:
Is it possible that creating multiple displays does limit the maximal movie count?

Because I get following message if I would like to create more then 4 `GLMovie`'s:

```
/Library/Caches/com.apple.xbs/Sources/AppleGVA/AppleGVA-10.1.17/Sources/Slices/Driver/AVD_loader.cpp: failed to get a service for display 4 
```

gohai, 2017-08-06:
@cansik This doesn't have anything to do with the issue at hand.

gohai, 2018-04-16:
I have decided to make the library Raspberry-Pi only, because I don't have enough time to deal with other platforms. Closing this bug for this reason.

closed 2018-04-16 by gohai