gohai/processing-glvideo #12: Certain video files don't work on OS X

gohai, 2016-08-27:
These two video files aren't currently working on OS X using the wip2 branch. They work from the command line on OS X, as well as on the Raspberry Pi using the 1.0.2 release.

[waters.zip](https://github.com/gohai/processing-glvideo/files/440669/waters.zip)

gohai, 2016-11-12:
This file also isn't working, albeit I've only tried with macOS.

[boulez.zip](https://github.com/gohai/processing-glvideo/files/587256/boulez.zip)

It might be that `texture-target: rectangle` is the culprit, see this thread [here](https://lists.freedesktop.org/archives/gstreamer-devel/2016-November/061489.html).

gohai, 2016-11-13:
Fixed by 9ec8f9ab5318b2a375af5f44ef853c156dd9b733

closed 2016-11-13 by gohai

DeqingSun, 2016-11-14:
Hi gohai.

I believed I has the same issue. How can I apply this fix?

gohai, 2016-11-14:
@DeqingSun Planning on releasing an updated version within a week or so.