gohai/processing-glvideo #17: Using glvideo with different Picam sensor modes

pikalu, 2016-09-14:
Dear gohai,

your library is just running fine on my Pi3, e.g. sketches like SimpleCapture just work fine. 

I know the library is at an very early state, but is there any possibility to change and use the different Picam sensor modes? If I interpret it correctly, the width and height of the video stram is currently fixed to 320x200.

Kind regards
pikalu

gohai, 2016-09-14:
What makes you believe it's always 320x200?

pikalu, 2016-09-14:
I just used println(video.width(), video.height()) in the setup function directly after play(). This returns always 320 200.
So maybe I used the wrong way to change resolution. I tried setting the camera to different modes by using raspivid beforehand and also tried to change the format by v4l2-ctl. Nothing worked.

Anyway, for this video stuff under Linux/Raspeberry I am a total beginner. So if this is a complete beginner mistake, do you have any additional advise for the right information source to start with?

gohai, 2016-09-14:
Which version of the GL Video library are you using? (the one currently available through the Contribution Manager or the most recently posted release candidate?)

What happens when you run `gst-launch-1.0 playbin uri=v4l2:///dev/video0` in the terminal? Same resolution as in Processing, or bigger?

pikalu, 2016-09-14:
I use the most recently posted release candidate. When executing your command, I get an livestream with the same resolution. In addition I receive a lot of warnings:

Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
libva info: VA-API version 0.36.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
libva info: VA-API version 0.36.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
libva info: VA-API version 0.36.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
libva info: VA-API version 0.36.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
libva info: VA-API version 0.36.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
libva info: VA-API version 0.36.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
Got context from element 'videosink-actual-sink-glimage': gst.gl.GLDisplay=context, gst.gl.GLDisplay=(GstGLDisplay)"\(GstGLDisplayEGL\)\ gldisplayegl0";

gohai, 2016-09-14:
Okay, in this case: it's really the default resolution of the V4L driver. Nothing we can do at this point until proper enumeration of video modes is implemented.

pikalu, 2016-09-14:
Thanks anyway for your support!

gohai, 2016-10-16:
@pikalu This is now implemented in branch `wip-list-devices` and will be part of the next release!

pikalu, 2016-10-18:
Sounds good, thanks!

gohai, 2016-11-12:
@pikalu This is in version 1.2 that was just released. Give it a try!

closed 2016-11-12 by gohai