gohai/processing-glvideo #19: use avfsvideosrc instead of qtkitvideosrc

codeanticode, 2016-10-11:
QTKit is being deprecated, so we should move over to this new source element:

https://cgit.freedesktop.org/gstreamer/gst-plugins-bad/tree/sys/applemedia/avfvideosrc.m?id=1.9.90

gohai, 2016-10-11:
This (w/ 1.8.3) works almost: https://github.com/gohai/processing-glvideo/tree/avfvideosrc

Getting black screen w/ `OpenGL error 1282 at bot endDraw(): invalid operation` (this is with Processing 3.2.1). Wrong pixel format? (I also tried adding ` ! videoscale ! videoconvert` to the pipeline, no joy.)

In the terminal, this works: `gst-launch-1.0 avfvideosrc ! glimagesink`

gohai, 2016-10-15:
Here is the full error

    com.jogamp.opengl.GLException: Thread[main-FPSAWTAnimator#00-Timer0,5,main] glGetError() returned the following error codes after a call to glBindTexture(<int> 0xDE1, <int> 0x4): GL_INVALID_OPERATION ( 1282 0x502), 
	at com.jogamp.opengl.DebugGL4bc.writeGLError(DebugGL4bc.java:31803)
	at com.jogamp.opengl.DebugGL4bc.glBindTexture(DebugGL4bc.java:730)
	at processing.opengl.PJOGL.bindTextureImpl(PJOGL.java:1423)
	at processing.opengl.PGL.bindTexture(PGL.java:3146)
	at processing.opengl.Texture.bind(Texture.java:680)
	at processing.opengl.PShader.setTexture(PShader.java:1338)
	at processing.opengl.PGraphicsOpenGL.flushPolys(PGraphicsOpenGL.java:2450)
	at processing.opengl.PGraphicsOpenGL.flush(PGraphicsOpenGL.java:2348)
	at processing.opengl.PGraphicsOpenGL.endDraw(PGraphicsOpenGL.java:1526)
	at processing.core.PApplet.handleDraw(PApplet.java:2430)

@codeanticode any idea?

codeanticode, 2016-10-15:
What gstreamer build are you using? Did yo try the 1.9.90 I uploaded a few days ago?

gohai, 2016-10-15:
With the 1.9.90 I don't get an error, but a white rectangle as output. (No joy w/ `videoscale ! videoconvert`) Does it work for you?

gohai, 2016-11-13:
Current code will automatically pick avfvideosrc over qtkitvideosrc with GStreamer versions 1.9 or higher. Believe 9ec8f9ab5318b2a375af5f44ef853c156dd9b733 should have file fixed the error I was seeing.

closed 2016-11-13 by gohai