gohai/processing-glvideo #26: Issues looping small, short video on Raspberry Pi

orgicus, 2017-03-12:
Hi,

I've just installed the `glvideo` library on a Raspberry Pi 2 running Processing 3.3
and tried to loop small (48x96), short(~10second) video, but after a few loops (about 3-4)
playback slows down and eventually freezes.

I'm running the exported SingleVideo example with the small video (instead launch1.mp4).
All I see in the output is this:
```bash
(java:4688): GStreamer-CRITICAL **: gst_tag_list_get_scope: assertion 'GST_IS_TAG_LIST (list)' failed

(java:4688): GStreamer-CRITICAL **: gst_tag_list_insert: assertion 'GST_IS_TAG_LIST (into)' failed
Final caps: video/x-raw(memory:GLMemory), format=(string)RGBA, width=(int)48, height=(int)96, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, texture-target=(string)2D
```
No other error messages.

I've also tried tweaking the generated launch script to use more memory (`Xmx128m`), but nothing changes. There still are 170MB of memory left while the app is running (but in a frozen state).

The video memory split is currently 128MB.

Any tips on how I could looping to work ?

Thank you,
George

gohai, 2017-05-24:
Might be the same as #4 - unfortunately I don't have the time to look into this currently.

orgicus, 2017-05-24:
Sounds a lot like it, sorry about the duplicate.

As temp workaround I used image sequences instead

Thank you

closed 2017-05-24 by orgicus