gohai/processing-glvideo #29: error logs : RANDR missing on display and more

rich-gg, 2017-05-05:
Hi

I just set up a spanking new raspberry 3 with its camera module V2
on the processing/raspian OS. 
In the SimpleCapture exemple, I had to replace video.start); by video.play();
but then, It works ! :D

but it give me all this in the console

Xlib:  extension "RANDR" missing on display ":10.0".
Devices:
[0] "mmal service 16.1"
Cannot parse capability: video/x-raw, format=(string)YUY2, width=(int)[ 32, 3280, 2 ], height=(int)[ 32, 2464, 2 ], pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string){ bt601 }, framerate=(fraction)[ 1/1, 90/1 ]
Cannot parse capability: video/x-raw, format=(string)UYVY, width=(int)[ 32, 3280, 2 ], height=(int)[ 32, 2464, 2 ], pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string){ bt601 }, framerate=(fraction)[ 1/1, 90/1 ]
Cannot parse capability: video/x-raw, format=(string)I420, width=(int)[ 32, 3280, 2 ], height=(int)[ 32, 2464, 2 ], pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string){ bt601 }, framerate=(fraction)[ 1/1, 90/1 ]
Cannot parse capability: video/x-raw, format=(string)YV12, width=(int)[ 32, 3280, 2 ], height=(int)[ 32, 2464, 2 ], pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string){ bt601 }, framerate=(fraction)[ 1/1, 90/1 ]
Cannot parse capability: image/jpeg, width=(int)[ 32, 3280, 2 ], height=(int)[ 32, 2464, 2 ], pixel-aspect-ratio=(fraction)1/1, colorimetry=(string){ 1:4:7:1 }, framerate=(fraction)[ 1/1, 90/1 ]
Cannot parse capability: image/jpeg, width=(int)[ 32, 3280, 2 ], height=(int)[ 32, 2464, 2 ], pixel-aspect-ratio=(fraction)1/1, colorimetry=(string){ bt601 }, framerate=(fraction)[ 1/1, 90/1 ]
Configs:
Cannot parse capability: video/x-raw, format=(string)BGRx, width=(int)[ 32, 3280, 2 ], height=(int)[ 32, 2464, 2 ], pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string){ 1:1:5:4 }, framerate=(fraction)[ 1/1, 90/1 ]
Cannot parse capability: video/x-raw, format=(string)BGR, width=(int)[ 32, 3280, 2 ], height=(int)[ 32, 2464, 2 ], pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string){ 1:1:5:4 }, framerate=(fraction)[ 1/1, 90/1 ]
Cannot parse capability: video/x-raw, format=(string)RGB, width=(int)[ 32, 3280, 2 ], height=(int)[ 32, 2464, 2 ], pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string){ sRGB }, framerate=(fraction)[ 1/1, 90/1 ]
Cannot parse capability: video/x-raw, format=(string)NV21, width=(int)[ 32, 3280, 2 ], height=(int)[ 32, 2464, 2 ], pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string){ bt601 }, framerate=(fraction)[ 1/1, 90/1 ]
Cannot parse capability: video/x-raw, format=(string)NV12, width=(int)[ 32, 3280, 2 ], height=(int)[ 32, 2464, 2 ], pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string){ bt601 }, framerate=(fraction)[ 1/1, 90/1 ]
Cannot parse capability: video/x-raw, format=(string)YVYU, width=(int)[ 32, 3280, 2 ], height=(int)[ 32, 2464, 2 ], pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string){ bt601 }, framerate=(fraction)[ 1/1, 90/1 ]
Cannot parse capability: video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, width=(int)[ 32, 3280, 2 ], height=(int)[ 32, 2464, 2 ], pixel-aspect-ratio=(fraction)1/1, colorimetry=(string){ bt601 }, framerate=(fraction)[ 1/1, 90/1 ]
Final caps: video/x-raw(memory:GLMemory), width=(int)320, height=(int)200, framerate=(fraction)90/1, format=(string)RGBA, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, texture-target=(string)2D

just so you know

gohai, 2017-05-05:
@rich-gg Thanks for the message!

The SimpleCapture example got changed as preparation for the next release, which will make use of the `start` function instead of `play` in order to be mimic the behavior of the regular Video plugin a bit closer. The other output that you're seeing is a known thing - need to get around to improving the capabilities parsing at some point (see #21).

closed 2017-05-05 by gohai

rich-gg, 2017-05-05:
Cool thx