gohai/processing-glvideo #32: MacOS error on loading video

cansik, 2017-06-09:
With version 1.1.2 your example **** is not running. There is a native error:

```java
(<unknown>:51500): GLib-GObject-WARNING **: cannot register existing type 'gchar'
**
GLib-GObject:ERROR:gvaluetypes.c:457:_g_value_types_init: assertion failed: (type == G_TYPE_CHAR)
Could not run the sketch (Target VM failed to initialize).
For more information, read revisions.txt and Help → Troubleshooting.

```

I'm running MacOS on a MPB 2016 with sierra (10.12.5 (16F73)) installed.

cansik, 2017-06-09:
Also happens with Version 1.2.1.

gohai, 2017-06-09:
Don't have a grip on what could be wrong here, sorry.

cansik, 2017-06-10:
Ok, but thanks for the fast answer. I'll investigate a bit. Maybe there is something wrong on my machine.

cansik, 2017-06-12:
So I figured out that the error is happening in this library, which is not the bundled one:

```
Application Specific Information:
/Library/Frameworks/GStreamer.framework/Versions/1.0/lib/libgobject-2.0.0.dylib
abort() called
```
So after removing the folder `GStreamer.framework` everything works as expected.

closed 2017-06-12 by cansik

gohai, 2017-06-12:
@cansik Thanks for this feedback! Do you know which version of the GStreamer.framework you had installed?

cansik, 2017-06-12:
@gohai The files are called `libgstreamer-1.0.0.dylib`, so I think it was version `1.0.0`.