gohai/processing-glvideo #33: No luck on OS X...

ffd8, 2017-06-13:
Maybe I'm doing something dumb or my setup is a little too old?
OS X 10.9.5, Processing 3.3.4, MBP 13" 2014 (Intel Iris 1536 MB)

With any example, I'm getting a -5 error (here with single video example):
``Debugging information: gsttypefindelement.c(1228): void gst_type_find_element_loop(GstPad *) (): /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstTypeFindElement:typefind:
streaming stopped, reason error (-5)
``

gohai, 2017-06-13:
@ffd8 Can you do me a favor and test versions 1.2.1 and 1.2, to check whether they work. They can be downloaded [here](https://github.com/gohai/processing-glvideo/releases).

Also, with the latest version, add the following line to your sketch so that it gets run before the constructor:

    GLVideo.enableDebug();

Copy the output and put it onto pastebin or similar.

ffd8, 2017-06-14:
@gohai Thanks for quick reply. Adding that debug code before constructor didn't change anything.. (if I understood correctly, placing it just before `video = new GLMovie(this, "launch1.mp4");`?

Was trying out 1.2.2 – just tried 1.2.1 and the program closes itself while trying to run.. two different error messages appeared, screen recorded to read 1st one:

``Glib-GObject: ERROR:gvaluetypes.c:455:_g_value_types_init: assertion failed: (type == G_TYPE_CHAR)
``

second error it gives while closing sketch:

``Could not run the sketch (Target VM failed to initialize).
For more information, read revisions.txt and Help → Troubleshooting.
``

Trying 1.2.0 = same issue...

Opened it in Processing 3.0 (versus 3.3.4) to see if that was the issue... and this time got more of an error message: https://pastebin.com/pKQ5jB0v

ffd8, 2017-06-14:
I also wondered if I am having the same [framework issue has here](https://github.com/gohai/processing-glvideo/issues/32) – So I tried removing that other framework (not sure from what program isn't there.. and hesitant to remove it) – didn't try a restart.. but simply moving it out of the folder (also 1.0) didn't solve the issue.

gohai, 2017-06-17:
@ffd8 This seems to be the culprit here

    dlopen(/Users/*******/Documents/processing_v3/libraries/glvideo/library/macosx/gstreamer-1.0/libgsttypefindfunctions.so, 2): Symbol not found: _fdopendir$INODE64
      Referenced from: /Users/*******/Documents/processing_v3/libraries/glvideo/library/macosx/gstreamer-1.0/../libgio-2.0.0.dylib
      Expected in: /usr/lib/libSystem.B.dylib

I searched for `fdopendir$INODE64` online, and while there was nothing regarding GStreamer per se, I did find various reports of users having issues running various open source software with your version of macOS (10.9). I would imagine that some part of the GStreamer might have an implicit dependency to a more recent version of macOS. We use the files from the "official" GStreamer build pretty much as-is, so I fear there is nothing really we can do if they end up targeting a more recent OS version.

gohai, 2018-04-16:
I have decided to make the library Raspberry-Pi only, because I don't have enough time to deal with other platforms. Closing this bug for this reason.

closed 2018-04-16 by gohai