gohai/processing-glvideo #35: Not possible to open more then 4 video files MacOS

cansik, 2017-08-06:
If I try to create more then 4 `GLMovie` I do get following error message. With the old video library & gstreamer 0.1 it was no problem to load multiple movies.

```
/Library/Caches/com.apple.xbs/Sources/AppleGVA/AppleGVA-10.1.17/Sources/Slices/Driver/AVD_loader.cpp: failed to get a service for display 4 
```

If you google for the error, you see that it is widely appearing in several applications. So it could be that it is not a problem of your library, but more of the Mac environment.

gohai, 2017-08-06:
@cansik Thanks for the report, I'll try to have a look into this.

gohai, 2018-04-16:
I have decided to make the library Raspberry-Pi only, because I don't have enough time to deal with other platforms. Closing this bug for this reason.

closed 2018-04-16 by gohai