gohai/processing-glvideo #41: cannot export  GLVideo application from raspbian/processing

aldobranti, 2018-02-03:
says there is no 32 GL Video library -- I thought that was the whole point with the embedded...

processing 3.3.5 as found in clean download from processing-raspbian

gohai, 2018-02-04:
@aldobranti What happens if you ignore this error? Will the application for linux-armv6hf still be created?

ChrisAhnberg, 2018-02-21:
I get the same error. The app will run, but no video. I've used your library many times (so thank you, thank you), but now I cannot get it to run at all. Even if I don't use the Upload to Pi tool and attempt to run the sketch straight from Processing on the Pi. 
[image]
[image]

gohai, 2018-02-22:
@ChrisAhnberg The error your are quoting is harmless, and can be ignored. Can you elaborate on the "does not run at all" part? Which platform are you trying to run the exported sketch on?

gohai, 2018-02-25:
@aldobranti @ChrisAhnberg Feel free to re-open this issue if you can provide more information - I was waiting for replies to my questions to you.