gohai/processing-glvideo #43: processing sketch fails quietly

aldobranti, 2018-02-24:
the below P3 sketch performs a continuous random scratch over two videos. The two videos are actually frame reversed versions of the other but as long as you have make them the same length, perhaps just a simple duplicate you should be able to run it
after as few as 7-800 or maybe some 10's of thousands of draw() calls it halts
I am running this on a RPI 3 and have set gui memory to 750M
I find nothing in the logs, if I run it under processing-java : nothing
? can you suggest why this sketch fails ?

`import gohai.glvideo.*; //<>// //<>//

class GLmovie_ext {
  GLMovie mov;
  float length;
};

GLmovie_ext[] m = new GLmovie_ext[2];
;

int now_showing;
boolean loaded = false;

void setup() { 
  //frameRate(50);
  size(500, 500, P2D);
  //fullScreen(P2D);
  //noCursor();
  m[0] = new GLmovie_ext();
  m[0].mov = new GLMovie(this, "clock_red.m4v", GLVideo.MUTE);  
  m[1] = new GLmovie_ext();
  m[1].mov = new GLMovie(this, "clock_green.m4v", GLVideo.MUTE);

  thread("loadMovies");
}

void loadMovies() {
  m[0].mov.loop();
  m[1].mov.loop();
  m[0].mov.volume(0);
  m[1].mov.volume(0);
  while (! m[0].mov.playing() || ! m[1].mov.playing()) delay(10);
  m[0].length = m[0].mov.duration();
  m[1].length = m[1].mov.duration();
  while ( m[0].length < 32 || m[1].length < 32 || abs( m[0].length - m[1].length ) > 0.1 ) {
    m[0].length = max( m[0].length, m[0].mov.duration());
    m[1].length = max( m[1].length, m[1].mov.duration());
  } 
  now_showing = 1;
  runTime = (m[0].length + m[1].length) / 2.0;
  loaded = true;
}

float cumulError = 0;
int drawCnt = 0;
float runTime;

void draw () {
  background(0);
  clear();
  if (!loaded) {
    textSize(10);
    fill(255);
    text("loading ...", width/2, height/2);
  } else {
    if (m[now_showing].mov.available()) m[now_showing].mov.read() ;   
    image( m[now_showing].mov, (width-m[now_showing].mov.width)/2, (height-m[now_showing].mov.height)/2);   
    int alternate = (now_showing+1) %2;
    if ( runTime < m[now_showing].mov.time() + m[alternate].mov.time() ) {
      float cutPoint = m[now_showing].mov.time();
      alternate = now_showing;
      now_showing = (now_showing+1) %2;
      cutPoint = random(0.99)*save+(runTime-save); // in the range from here to other's end point
      // now get alternate where we need it
      float distance = abs(cutPoint - save);
      float gap = runTime - 2*distance; // -1 .. -ve --> overshoot +ve means time in hand .. +1
      float correction = save + gap;
      while (correction > runTime) correction -= runTime;
      m[alternate].mov.jump( correction);
    }
  }
  stroke(color(#FF0000));
  fill(color(#FF0000));
  line(0, 10, width * (m[0].mov.time()/runTime), 10);
  if (now_showing == 0) {
    ellipse(0, 10, 10, 10);
  }
  stroke(color(#00FF00));
  fill(color(#00FF00));
  line(width * (1-m[1].mov.time()/runTime), 20, width, 20);
  if (now_showing == 1) {
    ellipse(width, 20, 10, 10);
  }
  drawCnt++;
  text(drawCnt, 40, 40);
  if (drawCnt % 1000 == 0) System.gc();
}

String fc(float F) {
  return nf( floor( 100.0*F/runTime));
}
`

gohai, 2018-02-25:
@aldobranti Can you upload the video files somewhere?

aldobranti, 2018-02-26:
please retrieve from https://aldobranti.eu/fosco_fornio/assets/applsnoranges_{red|green}.m4v
let me know if any problems

aldobranti, 2018-02-26:
that is not a directly accessible link, please make two separate down loads for the red and green files -- I was being lazy

aldobranti, 2018-02-27:
> JiRSIks5tYZ8OgaJpZM4SR-Gp> .
> ,"name":"GitHub"},"entity":{"external_key":"github/gohai/processing-glvideo","
> title":"gohai/processing-glvideo","subtitle":"GitHub
> repository","main_image_url":"https://cloud.githubusercontent.com/assets/14341
> 8/17495839/a5054eac-5d88-11e6-95fc-7290892c7bb5.png","avatar_image_url":"https
> ://cloud.githubusercontent.com/assets/143418/15842166/7c72db34-2c0b-11e6-9aed-
> b52498112777.png","action":{"name":"Open in
> GitHub","url":"https://github.com/gohai/processing-glvideo"}},"updates":{"snip
> pets":[{"icon":"PERSON","message":"@gohai in #43: @aldobranti Can you upload
> the video files somewhere?"}],"action":{"name":"View
> Issue","url":"https://github.com/gohai/processing-glvideo/issues/43#issuecomme
> nt-368329789"}}}

Did you find these ?
https://www.dropbox.com/s/bgpn2tbk99kljlp/applsnoranges_green.m4v?dl=0
https://www.dropbox.com/s/xs9tjmycz7husne/applsnoranges_red.m4v?dl=0

aldobranti, 2018-03-12:
to clarify -- the sketch freezes but remains as an active process. Looking at top I can see it is using processor power  and its memory allocation seems constant