gohai/processing-glvideo #45: SimpleCapture example failure

jansensan, 2018-03-13:
When running the `SimpleCapture` example, these errors are thrown:

````
MESA-LOADER: failed to retrieve device information
libGL error: MESA-LOADER: failed to retrieve device information
GLVideo requires the P2D or P3D renderer.

(java:769): GLib-CRITICAL **: g_error_free: assertion 'error != NULL' failed
````

- I am running this example on a Raspberry Pi, with a fresh install of Raspbian.
- The NoIR camera is connected properly, as running `raspistill -o /home/pi/Downloads/test.jpg` works properly.
- The entry `bcm2835_v4l2` is present in `/etc/modules`.

Any dependencies that could be missing or require an update in order for this to work?

The

gohai, 2018-03-13:
@jansensan Not sure exactly what is going on, but the `GLVideo requires the P2D or P3D renderer` line indicates that the library was unable to get the EGL context.

Did you by chance modify the `SimpleCapture` example to use the default renderer rather than the `P2D` or `P3D` one? Did you change the GL driver to be used in `raspbi-config`? Are you running Wayland?

If you want to test against a known-good distribution, please consider downloading our [Raspbian image with Processing pre-installed](https://github.com/processing/processing/releases/download/processing-0262-3.3.5/processing-3.3.5-linux-raspbian.zip).

jansensan, 2018-03-13:
No I didn't change anything in the example (and I rechecked after you asked, in case I did by mistake).

As for the GL driver, I didn't either. Which is sort of sad, because I wanted to use [PixelFlow](https://github.com/diwi/PixelFlow), but it requires GL.

Thanks for the suggestion about the Raspbian with Processing pre-installed, I will check that out.

gohai, 2018-03-13:
There _is_ GL out of the box - the reason I asked is because `raspbi-config` lets you choose between three different implementations ("Legacy", "Fake-KMS", and "KMS"). It's just that everything but the first (default) is currently not so well supported.

In case you're ever running the version of Raspbian that gave you this errors - could you test if regular `P3D` example sketches work for you?

jansensan, 2018-03-13:
Well, I have no real good explanation to give you...

- I ensured that the GL was set to legacy (which I remember setting as such, following the README instructions: "Do not enable the KMS OpenGL driver, but stick with the legacy one.")
- I ran example code provided on the [Processing P3D tutorial page](https://processing.org/tutorials/p3d/), which ran properly.
- I tried to run the SimpleCapture example again, and now it worked.

So I guess this issue could be closed, as it must have been some state of the Pi that didn't work properly. Sorry for the bother.

gohai, 2018-03-13:
@jansensan Hm, that's so curious. I don't have an answer to this either - anyway, thank you for reporting, and following up to the initial report! Feel free to re-open if this happens to you again. Best!

closed 2018-03-13 by gohai