gohai/processing-glvideo #46: OpenGL error

Johnreidsilver, 2018-03-24:
I'm unable to execute any glvideo examples. Simple P3D examples work.
Using Processing 3.3.7
more specs:
lsb_release -a
Description:	Ubuntu 17.10
Release:	17.10
Codename:	artful

java -version
openjdk version "1.8.0_151"
OpenJDK Runtime Environment (build 1.8.0_151-8u151-b12-0ubuntu0.17.10.2-b12)
OpenJDK 64-Bit Server VM (build 25.151-b12, mixed mode)

inxi -F
Graphics:  Card: Advanced Micro Devices [AMD/ATI] Richland [Radeon HD 8470D]
           Display Server: x11 (X.Org 1.19.5 ) driver: radeon
           Resolution: 1280x1024@85.02hz
           OpenGL: renderer: AMD ARUBA (DRM 2.50.0 / 4.13.0-37-generic, LLVM 5.0.0)

error trying to run singlevideo.pde for example:
The version of GStreamer on your system (1.12.3) does not match the version this library was compiled against (1.8). This is very likely to cause issues. Consider compiling this library yourself, using the sources from https://github.com/gohai/processing-glvideo.git

** (java:4215): CRITICAL **: gst_gl_handle_context_query: assertion 'display == NULL || DtsGetHWFeatures: Create File Failed
DtsGetHWFeatures: Create File Failed
Running DIL (3.22.0) Version
DtsDeviceOpen: Opening HW in mode 0
DtsDeviceOpen: Create File Failed
Final caps: video/x-raw(memory:GLMemory), width=(int)560, height=(int)406, framerate=(fraction)30000/1001, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, format=(string)RGBA, texture-target=(string)2D

How can I compile processing-glvideo against current GStreamer? I cloned the git cd into src/native and ran make and got fatal error: gst/gl/gl.h: No such file or directory

gohai, 2018-03-24:
Regarding `gst/gl/gl.h: No such file or directory`: you need to install the headers (`-dev`) for `gstreamer1.0-gst-plugins-bad`.

Regarding `Profile GL3bc is not available on X11GraphicsDevice`: this is an unrelated issue with `JOGL`, the OpenGL renderer Processing is using. Are you using the latest Processing release?

closed 2018-03-24 by gohai

Johnreidsilver, 2018-03-25:
I installed the same fresh download of processing-3.3.7-linux64.tgz in all computers tested. The Profile GL3bc error is on a Ubuntu 18.04 machine though. 

I can't seem to find the -dev package from ubuntu apt-get
First I did apt-get build-dep gstreamer1.0 , no -dev packages installed
Running apt-cache search gstreamer1.0* did find a -dev package but it was already installed: libgstreamer1.0-dev.
There are packages called debug, also installed bad-dbg, still same error trying to compile. No gl directory inside /usr/include/gstreamer-1.0/gst/

Are gstreamer's headers / -dev package available for download at some ppa or website?
Sorry for the noob questions.

Johnreidsilver, 2018-03-25:
Now the fun part. It works fine on this machine (it did throw an error once, but can't reproduce):

CPU:       Dual core AMD A6-6400K APU with Radeon HD Graphics (-MCP-) 
           cache: 2048 KB
           clock speeds: max: 3900 MHz 1: 2200 MHz 2: 2600 MHz
Graphics:  Card: Advanced Micro Devices [AMD/ATI] Richland [Radeon HD 8470D]
           Display Server: x11 (X.Org 1.19.5 ) driver: radeon
           Resolution: 1280x1024@85.02hz
           OpenGL: renderer: AMD ARUBA (DRM 2.50.0 / 4.13.0-37-generic, LLVM 5.0.0)

Linux a6 4.13.0-37-generic #42-Ubuntu SMP Wed Mar 7 14:13:23 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

Johnreidsilver, 2018-03-25:
Ok, so logging in on R7 PC with **xorg** instead of wayland makes everything work fine. 
So I guess this is related to wayland and not ati vs radeon driver.

Going to further test R3 PC with ubuntu 18.04 to see what's up with that error.