gohai/processing-glvideo #47: Render GLMovie in PGraphics buffer

TheodoreMichels, 2018-03-29:
Hi - first of all - thanks so much for all the work you've put into this project.

A quick overview of what I'm trying to accomplish:
Combine your GLVideo library with the Keystone Library to create some simple mapping utilities for the Pi.

Currently, I'm trying to render the GLMovie to an offscreen buffer using PGraphics, and then assigning that to a mapping surface using the keystone library's surface.render(buffer) function, but I'm getting a lot of errors, starting with:

```
GLVideo requires the P2D or P3D renderer.

(<unknown>:8891): GLib-CRITICAL **: g_error_free: assertion 'error != NULL' failed
java.lang.reflect.InvocationTargetException
```

The offscreen buffer is initialized with createGraphics(w, h, P2D)
This all works fine with standard Processing video.
I'd definitely be interested to know if there's a relatively simple solution to this issue - otherwise I know you've included a mapping example, so if all else fails I'll just build from that.

Thanks in advance,
Teddy

gohai, 2018-03-29:
Hi,

Is there a particular reason why you are using the default renderer for
your main sketch - and not P2D or P3D? (e.g. size(320, 240, P2D))

Note: there are also not one but two projection mapping examples included
with the GLVideo library. Those might work better than the keystone
library, at least on the Pi.

G

TheodoreMichels, 2018-03-30:
Thanks for getting back to me so quickly.
I am actually using the P3D renderer in my main sketch (keystone won't work without it) - part of why I thought that error was strange. If it helps, here's the repo for what I have so far:
https://github.com/TheodoreMichels/processing_mapper
Most stuff of interest is in the Surface.pde class.

This repo is actually currently using the standard video library - but there are no changes on the Pi version other than using GLMovie instead of Movie. 

And thanks for pointing out the other example - I hadn't tried the shader version yet - seems to perform better on the Pi.
Thanks again.

TheodoreMichels, 2018-03-30:
Hi again.
Sorry - I did what I should have done from the beginning and created a minimal test.
And in that test I was able to get the GLMovie to render to the PGraphics buffer and display on a keystone surface. So these two libraries seem to work perfectly well together.

I haven't pinpointed the problem exactly, but it seems that the issue had something to do with the fact that I was declaring and assigning the GLMovie variables on object creation, from a key press.

Putting the GLMovies in global scope seemed to fix the issue - I'm assuming that has something to do with the order of operations you mentioned.

And, for what it's worth, I'm running Jessie with default drivers.

If I find any more specific information about what particular implementation caused the problem, I'll be sure to post it, but otherwise everything seems to be working.
Thanks again for your insight.

closed 2018-03-30 by TheodoreMichels