gohai/processing-glvideo #53: Cannot link shader program

aegirfirefly, 2018-07-22:
Getting this error trying to use SimpleCapture with Raspberry Pi camera v2, so tried the SingleVideo example and got the same. It’s on raspbian, using built in Processing, I’ve run updates, checked all gstreamer packages etc. driver is loaded, have dev/video0. Only seen one mention of this error anywhere else (posted a couple of weeks ago) so drawing a blank on where to start.

```java.lang.RuntimeException: Cannot link shader program:
ERROR:PREPROCESSOR-1 (vertex shader, line 1) Preprocessor syntax error : garbage at end of preprocessor directive```

There’s a big list of errors thrown by PShader, PGraphics etc after this.

gohai, 2018-07-22:
Hi @aegirfirefly - that's unfortunately a bug that has crept into this particular release :(

Could either go back and use the [3.3.7](https://github.com/processing/processing/releases/tag/processing-0264-3.3.7), or you replace the `core.jar` that comes with your installation of Processing with [this file](https://sukzessiv.net/~gohai/p5-arm/mali/core.jar), or you wait for `3.3.7.2`, which should be available in a few hours.

aegirfirefly, 2018-07-22:
Ah! Thank you @gohai, that’s very much appreciated. I’ll update it tomorrow, thanks again.

RobotGrrl, 2018-07-24:
Can confirm the latest version works! Had the same error previously - updated to the latest version (clean install). Followed the steps to enable everything etc. SimpleCapture works and is displaying the Pi Cam! Thank you gohai and aegirfirefly :)

aegirfirefly, 2018-07-24:
Yup. Working for me now too.

closed 2018-07-24 by aegirfirefly