gohai/processing-glvideo #58: gohai does not let me start 2 cameras why?

FrankOcean11, 2019-04-24:
import gohai.glvideo.*;
GLCapture video,video2;

void setup() {
  size(320, 240, P2D);

  video = new GLCapture(this, devices[0]);// NOTE THIS CAM IS THE SAME AS devices[1] "USB2.0 PC Camera"
  video2 = new GLCapture(this, devices[1]);// NOTE THIS CAM IS THE SAME AS devices[0] "USB2.0 PC Camera"

  video.start();
  video2.start();
}

void draw() {
  background(0);
  if (video.available()) {
    video.read();
  }
  if (video2.available()) {
    video2.read();
  }
image(video, 0, 0, width, height);
image(video2, 0, 0, width/2, height/2);
}

Why wont processing or it's libraries do what i wrote above?

gohai, 2019-04-24:
Hi @FrankOcean11 - the issue here is that the device name is the same for both cameras. I am no longer working on this library, so I won't be able to do anything about it currently.

FrankOcean11, 2019-04-24:
right i know this but what i dont understand is why does this matter when they are both on 2 diff USB ports hence having 2 diff port address, well thanks thought i ask.