gohai/processing-glvideo #64: livecam example in openCV does not work with glvideo

FrankOcean11, 2019-07-17:
im using a raspberry pi 3B+ with a usb camera and cannot run example, i even tried the raspivid func in terminal and works wit raspberry pi camera module just fine but not in processing latest ide in opencv.

why is this happening? more importantly why are my post being closed before they get answered?

gohai, 2019-07-17:
Please use the Proceessing Forum for this type of troubleshooting. I am closing these bug reports because I am no longer maintaining this software. Sorry.

closed 2019-07-17 by gohai

FrankOcean11, 2019-07-17:
I did and they told me to bring it here, what is going on ?

gohai, 2019-07-17:
Sorry for the contradictory advise, @FrankOcean11. Can you point me to the discussion on the forum, so that I can inform people not to send users over here? Thanks in advance.